lsongdev/node-escpos #144: Printing QR-code, but only half appears

Xsmael, 2018-08-02:
Printing QR-code, but only half appears
[image]

Xsmael, 2018-08-06:
no, clue, but the HOIN HOP-H58 uses the same encoding than the one in the example, i  really dont understand

Psychopoulet, 2018-08-06:
I guess you have to specify the barcode's width

cybertim, 2018-08-06:
Tried to change the values using:
`.barcode('1234567', 'EAN8', { width: 50, height: 3 })`
but the printer will print text with only 'undefinedundefined' etc.. and i need to turn it on/off to get it working again.
also tried the legacy method:
`.barcode('1234567', 'EAN8', 50, 2, 'off', 'b')`
but this will not change anything (like, text is still printed below the small barcode which should be off).

My first impression is the barcode function has some bugs. Maybe later I have some time to dive into the code to see what is happening.

Btw. the default valus as per H58's programming docs is a default width of 50 and a height of 3.

Xsmael, 2018-08-06:
mmm, felt like peeping into the code too, but am under stress right now...

ograva, 2018-09-08:
I was actually using that part of the code as well, but could not center the QR Code. 
Alternatively, I just generated the QR code using another package called qrcode (npm) and just used the image function and it worked. 

Having problems though on my printer in that it cannot print consecutive times.

jchandr, 2019-10-12:
> I was actually using that part of the code as well, but could not center the QR Code.
> Alternatively, I just generated the QR code using another package called qrcode (npm) and just used the image [...]

can you please elaborate on the work around?

aureleduda, 2019-11-25:
you can use the qrimage function from this package, just add some options to set the margin and the size.

printer.align('ct').qrimage('https://youtu.be/W2ZmTMzN5NY', { type: 'png', margin:4, size:10}, function(err){ this.close();});

just look at qr-image options (https://github.com/alexeyten/qr-image) on which depends escpos