lsongdev/node-escpos #146: Images not printing

Xsmael, 2018-08-06:
I've tried the example at `node-escpos/examples/image.js`

```
const escpos = require('escpos');
const device  = new escpos.USB(0x456,0x808);
const options = { encoding: "GB18030" /* default */ };
// encoding is optional
 
const printer = new escpos.Printer(device, options); 
 
escpos.Image.load(__dirname + 'tux.png', function(image){

  device.open(function(){

    printer
    .align('ct')

    .image(image, 's8')
    .image(image, 'd8')
    .image(image, 's24')
    .image(image, 'd24')
    
    .raster(image)
    .raster(image, 'dw')
    .raster(image, 'dh')
    .raster(image, 'dwdh')
    .cut();
  });
});
```

But when i ruin this, absolutely nothing happens, and the cli (node) do not return

Psychopoulet, 2018-08-07:
__dirname + 'tux.png'
don't you this you missed something ? ^^

try __dirname + '/tux.png (like in the doc), or better and os agnostic : require('path').join(__dirname, 'tux.png');

Psychopoulet, 2018-08-07:
of course, if you don't have a picture named tux.png in the current directory you are in, there will be no picture to load...

Xsmael, 2018-08-07:
I downloaded the exact same tux.png and kept it in the same directory than my nodejs script.
I removed __dirname , because the image was in the same directory.

Psychopoulet, 2018-08-07:
__dirname + 'tux.png' => "directorytux.png"
__dirname + '/tux.png => "directory/tux.png"
require('path').join(__dirname, 'tux.png') => "directory/tux.png" | "directory\tux.png" (depend on your system)

if you remove the __dirname and you does not set the wanted directory in your terminal, you obviously will not load the file

Psychopoulet, 2018-08-07:
do you know the "cd" command (linux, mac & windows) ?

Xsmael, 2018-08-07:
yes

Xsmael, 2018-08-07:
`escpos.Image.load('./tux.png', function(image)`  this is  what i wrote and, by the way, when i put a wrong path (intentionnally)   i get this error:

> C:\node\thermalPrint\node_modules\escpos\printer.js:493
>     throw new TypeError('Only escpos.Image supported');
>     ^
> 
> TypeError: Only escpos.Image supported
>     at Printer.image (C:\node\thermalPrint\node_modules\escpos\printer.js:493:11)
>     at C:\node\thermalPrint\image.js:15:6
>     at Interface.<anonymous> (C:\node\thermalPrint\node_modules\escpos\adapter\usb.js:123:23)
>     at Device.<anonymous> (C:\node\thermalPrint\node_modules\usb\usb.js:239:6)
>

Psychopoulet, 2018-08-07:
that's it. use it to set the terminal into this directory if you don't use __dirname command.
if you use __dirname, use it with a separator beetween it and the file (path.join or simple "/" one)

Xsmael, 2018-08-07:
okay, this is the path to my project directory `C:\node\thermalPrint`
in there, i have image.js and tux.png, 

here is my CLI output:
[image]

as you can see, nothig happens, it doesnt return, and the printer doesnt react.

did i do anything wrong here ?

Xsmael, 2018-08-07:
also i notice that if instead of `./tux.png` i write a filename that doesn exist, i get this error: 
[image]

so ithink, that, maybe it finds the image file but something else bugs, cause in case the path is correct no errors, but no output

Xsmael, 2018-08-07:
damn it!    i just realilse, that  ` .close()` was missing sorry for all this, it will really reallly really be good if the example are working exmaple, it'll save us abit of time.
Thanks for all.

Psychopoulet, 2018-08-07:
thanks to you for pointing an example failure ! I reopen that thread

commit 38a79434 (2018-08-08): fix #146

https://github.com/song940/node-escpos/issues/146
closed 2018-08-08 by song940