lsongdev/node-escpos #147: Can i change the number of characters in a print line or print smaller characters

mfazevedo, 2018-08-07:
Right now my epson printer allows 42 characters on a single line. There is any way to increase this number? Even if the font size become smaller. 
In fact this is another thing that i dont know how to do.  In the code i just see size 1 and 2, but if i wanna print a smaller character? There is a way to acomplish that?

Psychopoulet, 2018-08-07:
it's a printer parameter, not an espos one
I don't think there is a way to bypass hardware limitations

mfazevedo, 2018-08-07:
In that case can i print a smaller font?

Psychopoulet, 2018-08-07:
yes, but you will be still limited to 42 characters (smaller or not)

mfazevedo, 2018-08-07:
can you post a example of a function that turn the font smaller? I already search and dont find one in the examples...

Psychopoulet, 2018-08-08:
in the documentation, you can find the "size" function

mfazevedo, 2018-08-08:
The number of characters in a single line is in fact a hardware setting. I changed this limitation direct on my printer setup.

But in the case os smaller characters i have no  further ideas. We have to print a coupon with some small fonts, and the smaller size (1,1) on library's documentation still too big for the purpose. 

Maybe this could be a future implementation that will help us a lot.

Psychopoulet, 2018-08-09:
"The number of characters in a single line is in fact a hardware setting.." => that's exactly what I said ><

for the characters limitations, I think you just have to deal with it.
Why do someone should try to set more characters in a line than the maximum ones supported by the printer ?