lsongdev/node-escpos #151: Problem with Cashdraw

vmaury, 2018-09-06:
I tried to let a cashdraw opening with cashdraw function.
But it was not running.
**I was obliged to change _.Cash_DRAWER object values in command.js line #60**
Origin :
CD_KICK_2: '\x1b\x70\x00', // Sends a pulse to pin 2 []
CD_KICK_5: '\x1b\x70\x01', // Sends a pulse to pin 5 []
My Changes
_.CASH_DRAWER = {
  CD_KICK_2: '\x1b\x70\x00\x19\xfa', // Sends a pulse to pin 2 []
  CD_KICK_5: '\x1b\x70\x01\x19\xfa', // Sends a pulse to pin 5 []

I found this values according to Epson Doc http://support.epostraders.co.uk/support-files/documents/27/Ya8-EpsonDrawerKick.pdf, page #4

Vehementis, 2019-08-19:
I can confirm this, came here to post the exact same thing.
\x19\xfa is needed or it does open it for me just every 2nd time i execute the command.
This two Codes represent the on and off time of the Signal, in this case 50ms ON and 500ms OFF

CajunDust, 2019-08-29:
Hello,
We also had the issue that the command did not execute from the initial call. But only after the 2nd call. More annoying was that this drawer command seemed to get "stuck" and also other calls (without the kick) did trigger the (unexpected) drawer open command ... (it looked like it was "queued" in front of the next print command...)
Thanks for this very helpfull fix !