lsongdev/node-escpos #152: Inconsistent flush of image

ograva, 2018-09-07:
I'm having trouble getting images to flush completely. It works the first time after the printer is reset.  
After that it just stops midway during the flush and does not proceed to close. 

When I run the same program again without resetting the printer I get the following error.

Can't close device with a pending request

Here is snippet of my code 

`device.open(function(){
      printer
      .font('A')
      .align('ct')
      .image(logo_image, 'd24')
      .flush()
      .text("ok")
      .image(qrimage, 'd24')
      .flush()
      .text("ok")
      .close(()=>{})
    });`

Psychopoulet, 2018-09-07:
you can use flush's callback to be sure the operation is done, before execute the next one

Psychopoulet, 2018-09-09:
printer.flush(() => {
// your code
});

song940, 2019-03-04:
Any questions?

closed 2019-03-04 by song940