lsongdev/node-escpos #158: Support ESC u (Check peripheral, cash drawer status)

mo-g, 2018-10-03:
https://reference.epson-biz.com/modules/ref_escpos/index.php?content_id=119

Would be really useful if node-escpos supported this esc/pos command, used on the TM-T88IV, V and VI among others.

Will have a look into writing a patch if no-one else gets there before I have time!

Psychopoulet, 2018-10-04:
"ESC u   [obsolete command]"

mo-g, 2018-10-04:
"Supported devices:
TM-m10
TM-m30
TM-T20
TM-T20II
TM-T70
TM-T70II
TM-T82II
TM-T88IV
TM-T88V
TM-T88VI
TM-U220"

Of which seven are available new from Epson at the time of writing.

mo-g, 2018-10-05:
Anyway, you're saying that if I was to submit a patch it would be rejected because this command is deprecated on future printers, correct?

Psychopoulet, 2018-10-08:
don't know, I submit patches, I don't accept them ^^

song940, 2019-06-25:
@mo-g pull-request welcome.

closed 2019-06-25 by song940