lsongdev/node-escpos #161: Can not Detect Printer over Virtual Com Port

ray-man, 2018-10-13:
In windows device manager under Universal Serial Bus device I see attached printer with name of 'Virtual Com Port' escpos is unable to communicate to with it. I've tried by providing `VID` and `PID` while defining usb device but doesn't work. `escpos.USB.findPrinter() ` doesn't find anything as well.

More over I've tried defining a serial port with device path but still no luck.

Any Suggestions ?

Psychopoulet, 2018-10-15:
did you install zadig driver ?

ray-man, 2018-10-19:
Yes zadig is installed and i've tried changing its drivers as well.

mfazevedo, 2018-10-30:
What's the printer model and manufacturer?

fdmatte, 2018-11-29:
I think im having the same problem, im using a Bematech MP-4200 TH and if i connect to the printer `via new escpos.Serial('COMPORT');` i can get it to work, but if i try to connect via USB with or without zadig it cannot find the printer.

I dont know if it helps but i think my printer does pass this statement on usb.js

`conf.bInterfaceClass === IFACE_CLASS.PRINTER`

Could the problem be that my printer is not identified by windows as a printer, instead it reckonizes it as a device?

mfazevedo, 2018-11-29:
With Bematech printers you can do like me and test the port configurations. In my case I created a constant called printerConnection that receives values like 'USB', 'COM1', 'COM2', etc.

The code:

if(printerConnection== "usb" || printerConnection== 'USB'){
          device = new escpos.USB(vid, pid);
}
else{
          device = new escpos.Serial(printerConnection);  //printerConnection='COMx'
}

You also don't need to install Zadic in that cases, just install the standart bematech printer driver and the "User software", that allow you to change the type  of input to "ESC/POS"

fdmatte, 2018-11-29:
Using this method, is there a way to auto detect which port is de printer using or do i have to hard-code it?

GuilhermeCouto, 2022-05-05:
did you solve it? I am facing the same problem.