lsongdev/node-escpos #164: Print problem with bematech MP 100S TH

mfazevedo, 2018-10-30:
Hi everyone!

I have this problem only on bematech mp100s th. When I send the print, some lines show a "5" char on begining. Another issue is that the number of columns is define with 42 chars and I need 48 chars.

In contat with bematech support, they tell me that the "5" char it's a problem with buffer cleaning, and the number of columns can be set with ESC/POS commands as well. The problem is: As we don't use the default bematech commands (http://bematechpartners.com.br/wp01/upload-files/downloads/windows/impressorasTermicas/Documentacoes/Manua-de-Programacao-MP-4200-TH.PDF) the support cannot help us. 

So in that case: Can I send the cleaning buffer command with this library, or set a number of columns?
If that is not possible, is a way to, using this library's commands, send the ESC/POS commands present on the linked manual?

Psychopoulet, 2018-10-31:
"Another issue is that the number of columns is define with 42 chars and I need 48 chars"
you cannot use 48 chars if your printer does not support it. it's an hardware configuration.

for the buffer, you have several ways to do it, like "printer.flush" method

Psychopoulet, 2018-11-02:
you certainely can use the printer's driver to configure it, but I don't know if we support that kind of commands