lsongdev/node-escpos #166: qrCode exceed the paper

Tina931010, 2018-11-05:
[image]
like this. The code image is not completed. The right part got lost.
It should be like 
[image]
How can I solve it?

Tina931010, 2018-11-05:
try this url 'http://im.gouyaoqu.com/qr/scan?qrCode=498bfad481ca43f5a4e99a1a40923533&type=DRUGSTORE_ORDER'

Tina931010, 2018-11-05:
device.open(function(){
    printer
      .qrimage('http://im.gouyaoqu.com/qr/scan?qrCode=498bfad481ca43f5a4e99a1a40923533&type=DRUGSTORE_ORDER', function(err){
        this.cut();
        this.close();
      })
  });

rafaelcardos0, 2018-12-15:
You can resize the QRCode image passing some options to qrimage method (second parameter).
Your code should looke like this:

`
device.open(function(){
	printer
		.qrimage('http://im.gouyaoqu.com/qr/scanqrCode=498bfad481ca43f5a4e99a1a40923533&type=DRUGSTORE_ORDER', { type: 'png', mode: 'dhdw', size: 4 }, function(err){
		this.cut();
		this.close();
	})
});
`

The default size is 5, try 4 or 3.

mladen-babic, 2019-12-29:
There is no "size" parameter in the options object !!!

Psychopoulet, 2020-01-02:
of course there is a "size" option !!
it's transfered to the "imageSync" method of the "qr-image" package

Psychopoulet, 2020-01-02:
there is a doc point, BTW.

closed 2020-01-02 by Psychopoulet

stevetsim, 2021-06-28:
> of course there is a "size" option !!
> it's transfered to the "imageSync" method of the "qr-image" package

`qrimage(content: string, options?: { type: string; mode: string }, callback?: (error: Error | null, printer?: Printer) => void): Printer;`

Actually, there is no "size" parameter in the options object as you can see above. So, how can I control the size of the qrcode?

ThayllanFelipe, 2022-06-15:
this doesn't work in typescript

abdulkholiqmahcoo-ux, 2025-09-30:
> [image] like this. The code image is not completed. The right part got lost. It should be like [image] How can I solve it?

catalandaniel983-cpu, 2026-08-16:
pls open my qrcode sg lisinse of sosia.