lsongdev/node-escpos #169: My printer prints qrimage without cutting paper.

flyFash, 2018-11-09:
When printing a qrimage, the printer sometimes cuts paper and sometimes does not cut it.When the paper is not cut, the method is executed, but the buffer.write method seems to have not been executed.
`this.printer.align("ct").qrimage(qrcodeUrl, function (err) {
       this.cut();
       this.close();
  });`
Everything is normal without printing the qrimage.

Psychopoulet, 2018-11-12:
use callbacks

rafaelcardos0, 2018-12-21:
What is your printer brand and model?

iYaroslav, 2019-05-19:
Same issue.
Printer VKP80II.
This printer has a command `eject`, it must be called immediately after the `cut`
Command: `GS e 3 0`

Psychopoulet, 2019-05-20:
probably because you didn't use "cut" method ? or correctly use callbacks ? without any context, it's a closed issue...

Psychopoulet, 2019-05-20:
probably because of a bad use of callback, or a missing "cut" method.
there is no context here...

Psychopoulet, 2019-05-20:
use callbacks, like
print("test", (err) => {
return err ? console.log(err) :  cut ((err) => {
console.log(err);
close();
})
});

iYaroslav, 2019-05-20:
``` js
/**
 * [function Cut paper]
 * @param  {[type]} part [description]
 * @return {[Printer]} printer  [the escpos printer instance]
 */
Printer.prototype.cut = function (part, feed) {
  this.feed(feed || 3);
  this.buffer.write(_.PAPER[
    part ? 'PAPER_PART_CUT' : 'PAPER_FULL_CUT'
  ]);
  return this;
};
```

Method `cut` not support callbacks
Or I do not understand something?

closed 2019-05-21 by Psychopoulet

rokobuljan, 2025-04-07:
After ages... Managed to cut (on NP 2411 D) uwing

```js

printer
  .align("ct")
  .text("TEST!")
  .raw(Buffer.from([0x1B, 0x64, 0x03, 0x1B, 0x6D]))
  .close();
```

Some other codes you could try:

```js
  .raw(Buffer.from([0x1D, 0x56, 0x42, 0x03])) // GS V B - Full cut with feed
  .raw(Buffer.from([0x1D, 0x56, 0x41, 0x03])) // GS V A - Partial cut with feed
  .raw(Buffer.from([0x1D, 0x56, 0x30])) // GS V 0 - Full cut
  .raw(Buffer.from([0x1D, 0x56, 0x31])) // GS V 1 - Partial cut   
  .raw(Buffer.from([0x1B, 0x64, 0x03, 0x1B, 0x6D])) // <<< I used this one for a partial cut
```

PS: don't be scared to try them all at once, and then see which one worked (if any at all).