lsongdev/node-escpos #173: `TypeError: Cannot read property 'getDeviceList' of null`

ozgursel, 2018-11-12:
`TypeError: Cannot read property 'getDeviceList' of null`

USB.findPrinter() is single method, not prototype and not execute USB() function like a class and thats why
usb library must be imported top of the code or you should add the code below every single methods

`if (!usb) { usb = require('usb'); }`

Or you need the figured out this another way with hotload.

_Originally posted by @ozgursel in https://github.com/song940/node-escpos/commit/52bc4e9ead64fbf5ecfbc3f323562ae610d6206f#commitcomment-31267432_

ozgursel, 2018-11-15:
So.. Are you using webpack?

Psychopoulet, 2018-11-15:
no. it's not a browser module, so...
I use gulp in the project which include this one, for linters, unit tests & code coverage

song940, 2019-03-04:
Any questions?

closed 2019-03-04 by song940