lsongdev/node-escpos #178: Cannot build package in directory with special characters

DaAwesomeP, 2018-12-03:
I cannot install escpos with a path like `/path/to/my project (has spaces and parentheses):
```
/bin/sh: -c: line 0: syntax error near unexpected token `('
/bin/sh: -c: line 0: `printf '%s\n' 'cmd_/path/to/my?project?(has?spaces?and?parentheses)/node_modules/usb/src/binding/usb_bindings.node := rm -rf "/path/to/my project (has spaces and parentheses)/node_modules/usb/src/binding/usb_bindings.node" && cp -af "Release/usb_bindings.node" "/path/to/my project (has spaces and parentheses)/node_modules/usb/src/binding/usb_bindings.node"' > ./Release/.deps//path/to/my?project?(has?spaces?and?parentheses)/node_modules/usb/src/binding/usb_bindings.node.d'
make: *** [action_after_build.target.mk:8: /path/to/my project (has spaces and parentheses)/node_modules/usb/src/binding/usb_bindings.node] Error 1
make: Leaving directory '/path/to/my project (has spaces and parentheses)/node_modules/usb/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:262:23)
gyp ERR! stack     at ChildProcess.emit (events.js:182:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:240:12)
gyp ERR! System Linux 4.12.14-lp150.12.25-default
gyp ERR! command "/usr/bin/node10" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/path/to/my project (has spaces and parentheses)/node_modules/usb/src/binding/usb_bindings.node" "--module_name=usb_bindings" "--module_path=/path/to/my project (has spaces and parentheses)/node_modules/usb/src/binding" "--napi_version=3" "--node_abi_napi=napi" "--napi_build_version=0" "--node_napi_label=node-v64"
gyp ERR! cwd /path/to/my project (has spaces and parentheses)/node_modules/usb
gyp ERR! node -v v10.13.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/usr/bin/node10 /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/path/to/my project (has spaces and parentheses)/node_modules/usb/src/binding/usb_bindings.node --module_name=usb_bindings --module_path=/path/to/my project (has spaces and parentheses)/node_modules/usb/src/binding --napi_version=3 --node_abi_napi=napi --napi_build_version=0 --node_napi_label=node-v64' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/path/to/my project (has spaces and parentheses)/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:182:13)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:962:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
node-pre-gyp ERR! System Linux 4.12.14-lp150.12.25-default
node-pre-gyp ERR! command "/usr/bin/node10" "/path/to/my project (has spaces and parentheses)/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /path/to/my project (has spaces and parentheses)/node_modules/usb
node-pre-gyp ERR! node -v v10.13.0
node-pre-gyp ERR! node-pre-gyp -v v0.11.0
node-pre-gyp ERR! not ok 
Failed to execute '/usr/bin/node10 /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/path/to/my project (has spaces and parentheses)/node_modules/usb/src/binding/usb_bindings.node --module_name=usb_bindings --module_path=/path/to/my project (has spaces and parentheses)/node_modules/usb/src/binding --napi_version=3 --node_abi_napi=napi --napi_build_version=0 --node_napi_label=node-v64' (1)
```
In some places it doesn't look like the spaces are escaped or that the spaces are replaced with question marks. I do not have this issue when building other native packages in directories with spaces and other characters in the path.

As a workaround, I built the package in a separate path and copied it to the new path.

Node v10.13.0
NPM v6.4.1
OpenSUSE Leap 15.1  x64
libudev-devel 234-lp150.20.9.1

Psychopoulet, 2018-12-04:
you should not use spaces in a path, whatever your project is...

DaAwesomeP, 2018-12-04:
Regardless, should the paths not be escaped/sanitized?

Psychopoulet, 2018-12-05:
it's a build error. not sure that is an escpos issue, maybe a node-gyp ?

DaAwesomeP, 2018-12-15:
Appears be an issue with the usb package, not escpos.

closed 2018-12-15 by DaAwesomeP

DaAwesomeP, 2018-12-15:
Please see https://github.com/tessel/node-usb/issues/287.