lsongdev/node-escpos #180: Use it in a browser

jarheadjoe, 2018-12-13:
Hi guys, someone try to use this package in browser using webpack/browserify? Or maybe it can't work in browser because it need to access the hardware?

Psychopoulet, 2018-12-13:
it's not designed to be used in a browser, but you can use the microserver embeded

jor3l, 2019-02-12:
The printer / comms do work in a browser, I think it is possible to create an adapter for both USB and BLE printers (I have done this but not using this as a package)

song940, 2019-03-04:
duplicate with https://github.com/song940/node-escpos/issues/180 https://github.com/song940/node-escpos/issues/150 https://github.com/song940/node-escpos/issues/196 https://github.com/song940/node-escpos/issues/76

closed 2019-03-04 by song940