lsongdev/node-escpos #181: Print problems with Promises

aroms, 2018-12-18:
Hello, I have to print different products, for different departments, each department with a different printer (so, different IP address). In this code I have three printers `array printers` from three different departments, and in each department a product assigned `array products`, (are related to the index, productos of `192.168.1.10` (printers index 0) ara products index 0: `[{ name: 'prod1 192.168.1.10' }, { name: 'prod2 192.168.1.10' }, { name: 'prod3 192.168.1.10' }]` and thus respectively):
```
  const printers = ['192.168.1.10', '192.168.1.11', '192.168.1.12'];
  const products = [
    [{ name: 'prod1 192.168.1.10' }, { name: 'prod2 192.168.1.10' }, { name: 'prod3 192.168.1.10' }],
    [{ name: 'prod1 192.168.1.11' }, { name: 'prod2 192.168.1.11' }, { name: 'prod3 192.168.1.11' }],
    [{ name: 'prod1 192.168.1.12' }, { name: 'prod2 192.168.1.12' }, { name: 'prod3 192.168.1.12' }],
  ];
```

The department head should know if all the products have been printed for each printer, and the way to do it is to use a `Promise.all`, basically, in the array `promisesPrinter` the promises returned by the printing () function are pushed, to be completed later with a Promise.all, which by means of a `.then()` can notify the head of department that not all the products have been printed, or that everything has good with `OK` or `ERROR`:
```
  const promisesPrinter = [];
  printers.forEach((printer, printerIndex) => {
    promisesPrinter.push(printing(printer, products[printerIndex]));
  });

  Promise.all(promisesPrinter).then((data) => {
    console.log(data); // [OK, OK, OK]
    console.log('OK');
  }, (error) => {
    console.log('ERROR');
  });
```
Here is the `printing()` function that returns a promise for Promise.all to work correctly:
```
function printing(printer, products) {
  console.log('printing - escpos.Network:', printer);
  const debugDevice = new escpos.Network(printer, 9100);
  const debugPrinter = new escpos.Printer(debugDevice);
  return new Promise((resolve, reject) => {
    debugDevice.open((error) => {
      console.log('return new Promise - error:', error);
      if (error) {
        reject(error);
      } else {
        resolve('OK');
      }
      debugPrinter
        .font('A')
        .align('LT')
        .style('BU')
        .size(1, 1)
        .text(`Printer: ${printer}`)
        .text('Products: ');
      products.forEach(product => debugPrinter.text(`${product.name}`));
      debugPrinter
        .text(' ')
        .text(' ')
        .cut()
        .close();
    });
  });
}
```

When executing it, it starts giving problems in `node_modules` `node-escpos` dependencies or like `ETIMEDOUT` (1 min 15 seconds to trigger this, `wow!`) combined with `ERR_STREAM_DESTROYED` or with `Maximum call stack size exceeded`:
```
node_modules/iconv-lite/lib/index.js:68
    var enc = (''+encoding).toLowerCase().replace(/[^0-9a-z]|:\d{4}$/g, "");
```

or also:
```
return new Promise - error: { [   connect ETIMEDOUT 192.168.1.10:9100 
  o net.js:1117 TCPConnectWrap.afterConnect [as oncomplete]
]
  errno: 'ETIMEDOUT',
  code: 'ETIMEDOUT',
  syscall: 'connect',
  address: '192.168.1.10',
  port: 9100 }
return new Promise - error: [   [ERR_STREAM_DESTROYED]: Cannot call write after a stream was destroyed 

  o _stream_writable.js:406 doWrite
  o _stream_writable.js:394 writeOrBuffer
  o _stream_writable.js:294 Socket.Writable.write
  o network.js:43 Network.write
[...]
[...]
```
or also:
```
node_modules/source-map/lib/util.js:174
    : normalize(aRoot.replace(/\/+$/, '') + '/' + aPath);
      ^
   Maximum call stack size exceeded 
  
  o util.js:174 Object.join
  o source-map-consumer.js:621 BasicSourceMapConsumer.SourceMapConsumer_originalPositionFor [as originalPositionFor]
  o source-map-support.js:199 mapSourcePosition
  o source-map-support.js:343 wrapCallSite
  o source-map-support.js:377 prepareStackTrace
[...]
[...]
```

:question: So what would be the way to solve it, any ideas? 

Certainly the documentation must be improved e.g with more examples, now it is too basic. :ok_hand:

Psychopoulet, 2018-12-19:
don't mess with the promises :

function printing(printer, products) {

  console.log('printing - escpos.Network:', printer);

  const debugDevice = new escpos.Network(printer, 9100);
  const debugPrinter = new escpos.Printer(debugDevice);

  return new Promise((resolve, reject) => {

    debugDevice.open((error) => {

      if (error) {
        console.log('return new Promise - error:', error);
        reject(error);
      } else {
        resolve();
      }

  }).then(() => {

      debugPrinter
        .font('A')
        .align('LT')
        .style('BU')
        .size(1, 1)
        .text(`Printer: ${printer}`)
        .text('Products: ');
      products.forEach(product => debugPrinter.text(`${product.name}`));

      return new Promise((resolve, reject) => {

        debugPrinter
          .text(' ')
          .text(' ')
          .cut()
          .close((err) => {
            return err ? reject(err) : resolve();
          });

      });
      
    });

  });

}

aroms, 2018-12-19:
Thanks for replying @Psychopoulet , but your modified version of the printing() function does not work for me, now the problem is:
```
debugDevice.open(...).then is not a function
```
*NOTE* - this is the printing() function as I have it now (it should be equal to the modified version of your comment, but I put it in case I have any erratum/typo):
```
function printing(printer, products) {
  console.log('printing - escpos.Network:', printer);

  const debugDevice = new escpos.Network(printer, 9100);
  const debugPrinter = new escpos.Printer(debugDevice);

  return new Promise((resolve, reject) => {
    debugDevice.open((error) => {
      if (error) {
        console.log('return new Promise - error:', error);
        reject(error);
      } else {
        resolve();
      }
    }).then(() => {
      debugPrinter
        .font('A')
        .align('LT')
        .style('BU')
        .size(1, 1)
        .text(`Printer: ${printer}`)
        .text('Products: ');
      products.forEach(product => debugPrinter.text(`${product.name}`));
      return new Promise((resolve, reject) => {
        debugPrinter
          .text(' ')
          .text(' ')
          .cut()
          .close((error) => {
            return error ? reject(error) : resolve();
          });
      });
    });
  });
}
```

Psychopoulet, 2018-12-20:
yep, sorry, typo problem. but you understood the logic, so...

nameszian, 2018-12-20:
Yes, I think I understand the problem, basically `open()` does not return a promise, and that's why you can not do `then()`, but what would be the solution? use `util.promisify` (from the official node 8+ util packages) to make a wrapper over `open()`? Is not it supposed that you already did it using `promiseify`?

nameszian, 2018-12-20:
With the following modified version of `printing()` it seems to work:

```
function printing(printer, products) {
  console.log('printing - escpos.Network:', printer);

  return new Promise((resolve, reject) => {
    const debugDevice = new escpos.Network(printer, 9100);
    debugDevice.open((err) => {
      if (err) return reject(err);
      return resolve(debugDevice);
    });
  }).then((device) => {
    const debugPrinter = new escpos.Printer(device);
    debugPrinter
      .font('A')
      .align('LT')
      .style('BU')
      .size(1, 1)
      .text(`Printer: ${printer}`)
      .text('Products: ');
    products.forEach(product => debugPrinter.text(`${product.name}`));
    debugPrinter
      .text(' ')
      .text(' ')
      .cut()
      .close();
  });
}
```

I will do tests these days, and if everything works correctly, I will make a PR to add it as an example (e.g. `examples/multipleNetworkPrinters.js`)

nameszian, 2018-12-21:
Any idea how to modify the default timeout that causes an `ETIMEDOUT`? Basically because I have tested it on several operating systems, and none of it drops below 1 minute (on my development computer it is always 1 minute and 15 seconds `wow!`), It's too slow, I need to trigger it faster, and I'm not the only one (e.g #49 ), any ideas?

https://github.com/song940/node-escpos/issues/49#issuecomment-431249110

Psychopoulet, 2018-12-21:
with the modified version, there will be no resolved promise. don't forget to return a promise after the "close" event

nameszian, 2018-12-21:
printing() must return a promise, it is promise.all who must resolve them.

jor3l, 2019-02-13:
Use [https://nodejs.org/api/net.html#net_socket_settimeout_timeout_callback](https://nodejs.org/api/net.html#net_socket_settimeout_timeout_callback))

closed 2019-02-14 by Psychopoulet

nameszian, 2019-02-14:
@jor3l It shows that you have no idea what you are talking about, that only serves to add more time, you should read everything before commenting...

jor3l, 2019-02-14:
based on your comment it seemed that you needed a way to set the socket timeout, the fact that it doesn't work for you doesn't mean I have no idea what I'm talking about, anyways since you're an expert please share the solution instead

nameszian, 2019-02-14:
I've already told you once and I'll repeat it again, you have to read well before commenting, here you can read what's going on, a little summarized (it's a previous comment):
> Any idea how to modify the default timeout that causes an `ETIMEDOUT`? Basically because I have tested it on several operating systems, and none of it drops below 1 minute (on my development computer it is always 1 minute and 15 seconds `wow!`), It's too slow, I need to trigger it faster, and I'm not the only one (e.g #49 ), any ideas?
> 
> [#49 (comment)](https://github.com/song940/node-escpos/issues/49#issuecomment-431249110)

ETIMEDOUT It seems to be a kernel issue that can not be customized (https://github.com/nodejs/help/issues/1686), it took more than a minute to give ETIMEDOUT, and your solution is to add more waiting time? It is simply a stupid response or made without reading what is asked, I simplify it, reduce the timeout ETIMEDOUT, not make it bigger.

Deberías leer antes de escribir, No seas Huevón y nos hagas perder el tiempo! (You should read before writing, Do not be a Huevón (fool), you make us waste our time!) 

Thanks for trying to help, but next time read before thinking of a solution, since if you do not read everything you have many chances that your solution is useless. 😸