lsongdev/node-escpos #182: how to print columns or table

netico14, 2018-12-22:
Hello, i need to print a table with 4 columns without lines, can i use this plugin for that?, thanks

Psychopoulet, 2019-01-02:
ok.
you can use this module (of course not in a front project), you just have to decide first how many characters the printer support, then divide in columns your data and autocomplete with blank spaces (before or after for alignement).

example : if the printer support 42 characters / line :

42 / 4 = 10,5 = 10 (you have to calculate this BEFORE and not for each line)

"test" "test2" "column" "column2" becomes
"test" (4 characters) => "test      "
"test2" (5 characters) => "test2     "
"column" (6 characters) => "column    "
"column2" (7 characters) => "column2   "

Psychopoulet, 2019-01-02:
... of course, github remove multiple blank spaces ><
so, "test" have (10 - 4 characters = 6) blank spaces,
etc...

closed 2019-01-02 by Psychopoulet

netico14, 2019-01-02:
Many thanks for the answer, the problem is that the text is dinamic, from
bbdd

Psychopoulet, 2019-01-03:
so, you have to compute it with break lines, but the process is still the same

netico14, 2019-01-03:
Great, thank you very much