lsongdev/node-escpos #184: Barcode options

czrm, 2019-01-10:
Hello, I was testing the latest master and noticed some weird things happening with the barcode options.
There is no example in the README with the options syntax, so I had to figure out the implementation using the info available, but, the documentation states that:
> "options.width" is a numeric value in the range between (1,255) Default: 64
> "options.height" is a numeric value in the range between (2,6) Default: 3

I was having trouble with this and reading through the source i found out that this is backwards, the width should be [1,2,3,4,5] as it is the key of the `BARCODE_WIDTH` object, and the height should be betwen 1~255.

Also I noticed that the line that says
> "options.position" is where to place the code around the bars, could be one of the following values:
> * ABOVE
> * BELOW
> * BOTH
> * OFF

in the source code was actually expecting:
* OFF
* ABV
* BLW
* BTH

I like to know if my findings are correct, and what is the correct form of the barcode options.
If there is an error, being documentation or source, I would be more than happy to contribute by fixing it, but right now I don't know what is the expected form

Cheers

EverettRhodes, 2019-04-18:
Thanks for posting this. I was struggling with the options since the README is wrong.

dudifreire, 2021-01-26:
what's the current sintax ? i'm getting "width is not defined" or "option is not a function"

czrml, 2021-01-26:
Hi @dudifreire, 

It has been some years since I've worked with node and looked into this.
I couldn't find my original working code to paste it here but I hope I can be of some assistance

This issue was related to [this](https://github.com/song940/node-escpos/tree/v3/packages/printer) package's [README.md](https://github.com/song940/node-escpos/blob/v3/packages/printer/README.md)

The example shown there could use some improvement but looking at the source code we can find this:

```js
Printer.prototype.barcode = function (code, type, options) {
  options = options || {};
  var width, height, position, font, includeParity;
  // Backward compatibility
  width = arguments[2];
  if (typeof width === 'string' || typeof width === 'number') {
    width = arguments[2];
    height = arguments[3];
    position = arguments[4];
    font = arguments[5];
  } else {
    width = options.width;
    height = options.height;
    position = options.position;
    font = options.font;
    includeParity = options.includeParity !== false; // true by default
  }
}
```

From that I assume you could call the function passing the width as parameter as so:
```js
printer
  .barcode('1234567', 'EAN8', {width: 3, height: 64})

// OR

let options = {
  width: 3,
  height: 64
}

printer
  .barcode('1234567', 'EAN8', options)
```

Notice the content of this issue, in the documentation it was stated that `options.width` should be an integer between 1 and 255 but the code in fact [expects a number between 1 and 5](https://github.com/song940/node-escpos/blob/d3e9c1550d0f105da6361cd18e63cc55d82874f1/packages/printer/index.js#L576)

I've hope this works and can be of some assistance to you, It has been years since I've worked with JS so the code that I wrote may have some syntax errors and such.

This library doesn't seem to be maintained anymore, but this issue is still relevant and it seems that was never fixed so I'll leave it open