lsongdev/node-escpos #190: Can set text alignment or create table layout

siteslave, 2019-02-03:
Can set text alignment like this.

```
printer.leftRight("Left", "Right"); 
printer.table(["One", "Two", "Three"]);

printer.tableCustom([ 
  { text:"Left", align:"LEFT", width:0.5 },
  { text:"Center", align:"CENTER", width:0.25, bold:true },
  { text:"Right", align:"RIGHT", width:0.25 }
]);

```

demo code from [https://www.npmjs.com/package/node-thermal-printer](https://www.npmjs.com/package/node-thermal-printer)

Psychopoulet, 2019-02-04:
you have all the tools to wrap it anyway... that's I do, with XML parser to create a basic API
you cannot create "table" method" without knowing the printer's max characters/line

song940, 2019-03-04:
Any questions?

closed 2019-03-04 by song940

brkrtp, 2019-06-21:
> printer's

Can you send example code?

Psychopoulet, 2019-06-24:
simple html parser and some loops for autocomplete columns with spaces, with a predetermined width (42 chars by default)