lsongdev/node-escpos #193: How to set instructions to POS

TomaIvanovTomov, 2019-02-11:
Hi. Mi POS terminal has instructions ( HEX codes ) in its documentation. The question is: How to send those instruction ? I mean with wich method ? Lets say I want to roll a little papper. The instruction for this is "2C" hex number. Is it possible? Because with the default example
const escpos = require('escpos');

```
// Select the adapter based on your printer type
const device  = new escpos.USB();
// const device  = new escpos.Network('localhost');
// const device  = new escpos.Serial('/dev/usb/lp0');

const options = { encoding: "GB18030" /* default */ }
// encoding is optional

const printer = new escpos.Printer(device, options);

device.open(function(){
  printer
  .font('a')
  .align('ct')
  .style('bu')
  .size(1, 1)
  .text('The quick brown fox jumps over the lazy dog')
  .text('敏捷的棕色狐狸跳过懒狗')
  .barcode('1234567', 'EAN8')
  .qrimage('https://github.com/song940/node-escpos', function(err){
    this.cut();
    this.close();
  });
});
```
, wich is shown in the npm-escpos documentatiton, the printer doesn't respond. I tried to put the code in `text` method - `.text("2C")` but it also doesn't work . Any idea ? Thank you very much!

Psychopoulet, 2019-02-12:
you can create your own adapter and redirect the flow on your terminal

song940, 2019-03-04:
Any questions?

closed 2019-03-04 by song940