lsongdev/node-escpos #197: Does this package works with react native?

shubhamdeol, 2019-03-25:


song940, 2019-03-28:
This library is not designed for browser stack, but you can use network adapter to connect printer.

jor3l, 2019-03-28:
worth to mention, you can use a usb printer with it by creating a custom adapter as well

Psychopoulet, 2019-03-29:
he ask a question about react NATIVE, which is not designed for browser but for mobile.
IDK if it works with it, but I guess it should (with network adapter for example) ? should be tested.

jor3l, 2019-03-29:
My bad, not an easy test imo but it could work, considering many of the dependencies get to work with react native

shubhamdeol, 2019-03-29:
I really don't have any idea about how network adapter works with react native.  Please someone explain widely.

Psychopoulet, 2019-04-01:
it's just a basic socket wrapper for nodejs, so it should works
https://github.com/song940/node-escpos/blob/master/adapter/network.js

closed 2019-06-25 by song940