lsongdev/node-escpos #199: Printer is installed and working but I always get LIBUSB_ERROR_NOT_FOUND

jabawack81, 2019-03-26:
this is my test code:
```
const escpos = require('escpos');
const usb = require("usb");

const device0 = usb.findByIds(0x0AA7, 0x0304);

console.log(device0);

// Select the adapter based on your printer type
const device  = new escpos.USB(0x0AA7, 0x0304);
// const device  = new escpos.Network('localhost');
// const device  = new escpos.Serial('/dev/usb/lp0');

const options = { encoding: "GB18030" /* default */ }
// encoding is optional

const printer = new escpos.Printer(device, options);

device.open(function(){
  printer
  .font('a')
  .align('ct')
  .style('bu')
  .size(1, 1)
  .text('The quick brown fox jumps over the lazy dog')
  .text('敏捷的棕色狐狸跳过懒狗')
  .barcode('1234567', 'EAN8')
  .qrimage('https://github.com/song940/node-escpos', function(err){
    this.cut();
    this.close();
  });
});
```

and this is the output:

```
> printer@1.0.0 start C:\printer
> node index.js

Device {
  busNumber: 3,
  deviceAddress: 6,
  deviceDescriptor:
   { bLength: 18,
     bDescriptorType: 1,
     bcdUSB: 512,
     bDeviceClass: 0,
     bDeviceSubClass: 0,
     bDeviceProtocol: 0,
     bMaxPacketSize0: 64,
     idVendor: 2727,
     idProduct: 772,
     bcdDevice: 264,
     iManufacturer: 1,
     iProduct: 2,
     iSerialNumber: 3,
     bNumConfigurations: 1 },
  portNumbers: [ 7, 4 ] }
C:\printer\node_modules\usb\usb.js:38
        this.__open()
             ^

Error: LIBUSB_ERROR_NOT_FOUND
    at Device.usb.Device.open (C:\printer\node_modules\usb\usb.js:38:7)
    at USB.open (C:\printer\node_modules\escpos\adapter\usb.js:107:15)
    at Object.<anonymous> (C:\printer\index.js:20:8)
    at Module._compile (internal/modules/cjs/loader.js:701:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:712:10)
    at Module.load (internal/modules/cjs/loader.js:600:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:539:12)
    at Function.Module._load (internal/modules/cjs/loader.js:531:3)
    at Function.Module.runMain (internal/modules/cjs/loader.js:754:12)
    at startup (internal/bootstrap/node.js:283:19)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! printer@1.0.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the printer@1.0.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\_logs\2019-03-26T16_38_07_582Z-debug.log
```

so if I'm right `node-usb` can see the printer so I'm not sure why escpos can't

Psychopoulet, 2019-03-27:
did you install zadig driver ?

jabawack81, 2019-03-27:
yes I tried but they made everything worst because the printer wasn't recognized by windows and the same code was crashing with device not supported

jabawack81, 2019-03-27:
Reinstalling a couple of times the zadig driver made the trick I'm not sure why but I'm going to close this issue feel free to delete it

closed 2019-03-27 by jabawack81