lsongdev/node-escpos #203: The "images.js" in "examples" can not run.

ycy0312, 2019-04-08:
I tried the "images.js" in "examples" but it could not run successfully.
[image]

Psychopoulet, 2019-04-08:
... can we have a little context, please ?
OS ? node version ? code used ? etc...

mbrodt, 2019-04-08:
Not OP, but I'm having the same issue. Running node v10.15.3 on macOS Mojave 10.14.3.

Code here:

`const device = new escpos.USB()
  const options = { encoding: 'GB18030' /* default */ }
  const printer = new escpos.Printer(device, options)

  const tux = path.join(__dirname, 'tux.png')
  escpos.Image.load(tux, function(image) {
    device.open(function() {
      printer
        .image(image, 's8')
        .image(image, 'd8')
        .image(image, 's24')
        .image(image, 'd24')
        .raster(image)
        .raster(image, 'dw')
        .raster(image, 'dh')
        .raster(image, 'dwdh')

        .cut()
        .close()
    })
  })`

The tux is in the right place, and the error is the exact same as OP's.

Any help is appreciated!

ycy0312, 2019-04-09:
Sorry...My fault.
OS：windows 10
node version：v8.9.3
Code is same as the example.
[image]

jor3l, 2019-04-09:
Can you guys try that example without the chaining? ( put a printer on each line)

mbrodt, 2019-04-10:
Tried that, same issue. Also tried a simple printer.image(image, 's8').cut().close(), but the error stays the same.

song940, 2019-04-10:
Because [printer#image](https://github.com/song940/node-escpos/blob/master/printer.js#L491) is async function, so you can not chaining.

mbrodt, 2019-04-10:
That does it. Thanks a lot guys!

closed 2019-04-16 by song940