lsongdev/node-escpos #205: Print on another port other than the 9100

nmorales1991, 2019-04-23:
Hi. Can I print on another port other than the 9100?

song940, 2019-04-24:
```js
const device  = new escpos.Network('localhost', 9200); // <-- You can change the port
const printer = new escpos.Printer(device);
```

closed 2019-04-24 by song940

nmorales1991, 2019-04-24:
> ```js
> const device  = new escpos.Network('localhost', 9200); // <-- You can change the port
> const printer = new escpos.Printer(device);
> ```

But do you have to configure something else? since with any other port I get an error

Psychopoulet, 2019-04-25:
the IP ? I guess the printer is not on localhost...

nmorales1991, 2019-04-25:
> the IP ? I guess the printer is not on localhost...

Is a Network Connection to other ip, is not localhost.
Then if it is to another ip, can not be with another port?

Psychopoulet, 2019-04-26:
... what ?

nmorales1991, 2019-04-26:
> ... what ?

The printer is not localhost. Is a network connection to other ip . This works only with the port 9100

`const device  = new escpos.Network('ip printer', 9100);`

But with any other port it does not work

Psychopoulet, 2019-05-01:
you does not have control on port's change, so you can change it without problem.
If the changed port does not work, you should try to telnet it because you probably try an inexistant IP/port combinaison