lsongdev/node-escpos #206: Example for Screen device

Tauri28, 2019-04-29:
I get "CONNECTED" and "CLOSED" but screen remains blank. Am I missing something?

```
import * as escpos from "escpos";
const device = new escpos.USB(0x04B8, 0x0202);
const screen = new escpos.Screen(device);
device.open((err) => {
	if (err) throw err;
	console.log("CONNECTED");
	screen.moveHome();
	screen.text("Hello!", "utf8");
	screen.close(() => {
		console.log("CLOSED");
	});
});
```

song940, 2019-04-30:
Maybe @atiertant can help you.

nicolasbadia, 2019-04-30:
Same problem here. Nothing is happening.

atiertant, 2019-05-02:
@Tauri28 i tryed on Linux with oxhoo AF240U.
it required configure device to esc/pos and use minicom to use serial port.

closed 2019-06-25 by song940

nicolasbadia, 2019-06-25:
For the record, after connecting the display directly to the computer and getting the ID with `lsusb`, everything worked.

`lsusb`:
```
Bus 001 Device 004: ID 1504:008c
```

JS:
```
const device = new escpos.USB(0x1504, 0x008c);
const screen = new escpos.Screen(device);
```