lsongdev/node-escpos #210: Multiple printers

diegomachado1, 2019-05-19:
Hey guys..

I'm trying to use this module with multiples printers. But some codes aren't working.

So, I read ([#23](https://github.com/song940/node-escpos/pull/23)), ([#173](https://github.com/song940/node-escpos/issues/173)).

I tried to follow the [multiple.js](https://github.com/song940/node-escpos/blob/master/examples/multiple.js) example. But the code for it self doesn't work for some reasons:

1. The code use `device.open()`, and `device` isn't defined
2. `escpos.USB.findPrinter()` returns this error:
```
TypeError: Cannot read property 'getDeviceList' of null
    at Function.USB.findPrinter
```
I can get `findPrinter()` working if I call `new escpos.USB()` before it. But I don't know why..

So, my final code was link this:
```
let device = new escpos.USB()
const devices = escpos.USB.findPrinter();

const printer = new escpos.Printer([
    devices[0],
    devices[1]
]);

device.open(function(){
    printer
    .font('a')
    .align('ct')
    .style('bu')
    .size(1, 1)
    .text('The quick brown fox jumps over the lazy dog')
    .cut()
    .close();
});
```
And I get this error when I try to execute it:
```
TypeError: this.adapter.write is not a function
    at Printer.flush
```

I don't know if I've misunderstood something on the docs (sorry if I did) or if the docs are wrong. But how can I get it working?

And, what is the best approach to manage different printing formats with this logic? The printers stay on different places and one need to receive less information then another. 
If I have two printers with the same VID and PID I can't know what printer should receive less information, right?

daniellizik, 2019-05-28:
It's an issue with your driver or hardware configuration

diegomachado1, 2019-05-28:
I tried this both on mac and windows. I think it's difficult to have the same problem on the two pc. But how can I have sure about it? Has some tip to fix it?

rssk, 2019-07-26:
This fails on my mac as well

daniellizik, 2019-07-26:
This has failed for me in the past if my epson TM-T88* (V or VI) is not configured to vendor class. Sometimes it cannot be discovered if it set to printer class.

rssk, 2019-07-26:
Interesting, I've got everything working now doing it manually on a brother wide format. There are some minor command differences but wasn't hard to mess with, thanks for the library!

smailmalki, 2020-03-31:
my answer is (very) late
You have to do :
`const escpos = require('escpos');
escpos.USB = require('escpos-usb');
var devices = escpos.USB.findPrinter();

devices.forEach(function(el) { 
    let device = new escpos.USB(el)
    const printer = new escpos.Printer(device);
    device.open(function(){
        printer
        .font('a')
        .align('ct')
        .style('bu')
        .size(1, 1)
        .text('The quick brown fox jumps over the lazy dog')
        .cut()
        .close()
    });
})`

Hope it helps you

closed 2020-05-12 by song940

strife-cloud, 2022-10-26:
Hi, to add some details about how to chose specificaly one printer, I look on object type Device the property portNumbers that is the only distinct property in the object
`const devices = escpos.USB.findPrinter();
              deviceChosen = devices.find(printerUSB => {
                return JSON.stringify(printerUSB.portNumbers) === printParam.address;
              });`

Another improve that is looked for serial number like with the usb package:
https://www.npmjs.com/package/usb look for "Find legacy device by SerialNumber"

dohooo, 2022-10-26:
> Hi, to add some details about how to chose specificaly one printer, I look on object type Device the property portNumbers that is the only distinct property in the object
> 
> `const devices = escpos.USB.findPrinter();
> 
>               deviceChosen = devices.find(printerUSB => {
> 
>                 return JSON.stringify(printerUSB.portNumbers) === printParam.address;
> 
>               });`
> 
> 
> 
> Another improve that is looked for serial number like with the usb package:
> 
> https://www.npmjs.com/package/usb look for "Find legacy device by SerialNumber" so we can chose specifically one printer

#412