lsongdev/node-escpos #214: Can't import module in electron 4.1.1

MedAmineBenyettou, 2019-06-16:
Am trying to use this module with electron v4.1.1 and node v10.16.0 but am getting this error :

ode_modules\@serialport\bindings\build\Release\bindings.node' was compiled against a different Node.js version using NODE_MODULE_VERSION 64. This version of Node.js requires NODE_MODULE_VERSION 69.

diegomachado1, 2019-06-16:
Electron 4.1.1 require Node 10.11.0.
You can use [NVM](https://github.com/nvm-sh/nvm) to manage your node environment
Install the required version, delete your node_modules folder and rum **npm install** again

MedAmineBenyettou, 2019-06-16:
Well, I can't downgrade my electron + node for just 1 module, Am at the final stage of my project and I can't afford the time to do such a dangerous thing.

I never had such an error before and I think it's only related to the version of @serialport\bindings so maybe updating that file will help solve the probleme ?

diegomachado1, 2019-06-17:
Ok, try this:
```
npm install electron-rebuild --save-dev

// execute the command this way:
./node_modules/.bin/electron-rebuild

// and then run your application like this:
./node_modules/.bin/electron .
````
And every time you add a new package, you'll need to run **electron-rebuild**

MedAmineBenyettou, 2019-06-18:
That's fixed the including probleme, but then when I do : 

    ` const device = new escpos.USB();`

It just ... Crashes : no error no nothing, it just says that the DevTools was disconnected from this page.

closed 2020-06-16 by song940