lsongdev/node-escpos #220: DeprecationWarning: Buffer() is deprecated due to security and usability issues

qaoo8, 2019-08-08:
I am using Bluetooth to print the demo of index.js in the example folder; it works well with a print out but got this Warning after the run command: node index.js; but if I run the command again, it print out nothing; the command was skipped 

**"(node:7075) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead."**

I can not fix this issue by myself, Please advice!

Psychopoulet, 2019-08-10:
weird...
node version ? os ? stack ?

lbarthon, 2019-08-13:
The version you are using might not be the last one, since Buffer() isn't used in the code.
Are you using 2.5.1 ?

qaoo8, 2019-08-26:
I have two Bluetooth printers connected to my computer for testing, both are the same problem; 
What I had done are clone the repo and add my device and run the script; here are my settings: 
I am using ZSH in VS CODE insider;
[image]