lsongdev/node-escpos #221: We should move adapters out of this repository. Then we will be able to use on electorn or web browser.

mofelee, 2019-08-13:
like this repo: https://github.com/mofelee/escpos-without-adapter

electron network adapter on renderer process:
```js
const net = window.require('net')
```

Psychopoulet, 2019-08-18:
I think we should optionaly load the adapters, but keep it in the package

mofelee, 2019-08-18:
> I think we should optionaly load the adapters, but keep it in the package

But if adapter's dependency in package.json will cause some problem, like usb or serialport is not woking on electron, but it will try to build those dependencies.

removed from package.json will crash app when dependency missing.

Psychopoulet, 2019-08-19:
with a optional load, there is no problem. the problem only occurs when you use the "require" method, that's why I proposed a dynamic load for usb adapter on the first place.
So, if you keep it in the module but only load it when it's wanted, you will not have this issue in any case, but if you modularize it, it will be called and cause the crash anyway if you put in in the package.json as a dep

Mwangangi, 2019-10-16:
@mofelee Does your fork function well on the browser after removing the adapters? If yes, can it print via USB

mofelee, 2019-10-28:
> @mofelee Does your fork function well on the browser after removing the adapters? If yes, can it print via USB

My printer is network printer, electron can direct use the "net" module on browser. but we need require net module via `window.require('net')`

closed 2020-02-24 by song940