lsongdev/node-escpos #223: .table is not a function

elixiao, 2019-08-26:
When I run the code in REAMDE

```js
const escpos = require('escpos')
const device = new escpos.USB()
const options = { encoding: 'GB18030' }
const printer = new escpos.Printer(device, options)
device.open(function() {
  printer
    .font('a')
    .align('ct')
    .style('bu')
    .size(1, 1)
    .text('The quick brown fox jumps over the lazy dog')
    .text('敏捷的棕色狐狸跳过懒狗')
    .barcode('1234567', 'EAN8')
    .table(['One', 'Two', 'Three'])
    .tableCustom([
      { text: 'Left', align: 'LEFT', width: 0.33 },
      { text: 'Center', align: 'CENTER', width: 0.33 },
      { text: 'Right', align: 'RIGHT', width: 0.33 }
    ])
    .qrimage('https://github.com/song940/node-escpos', function(err) {
      this.cut()
      this.close()
    })
})
```

The following error occurs:

```sh
      .table(['One', 'Two', 'Three'])
       ^

TypeError: printer.font(...).align(...).style(...).size(...).text(...).text(...).barcode(...).table is not a function
```

The code could run when I remove `table` and `tableCustom` function

```js
   .table(['One', 'Two', 'Three'])
   .tableCustom([
      { text: 'Left', align: 'LEFT', width: 0.33 },
      { text: 'Center', align: 'CENTER', width: 0.33 },
      { text: 'Right', align: 'RIGHT', width: 0.33 }
    ])
```

Has this feature not implemented yet?

elixiao, 2019-08-26:
Update to latest version fix the problem. But the table format seems not correct. Shouldn't it be on the same line?

[image]

elixiao, 2019-08-26:
Fix it by replace 48 by 32 in [print.js line 154](https://github.com/song940/node-escpos/blob/master/printer.js#L154) and [line 186](https://github.com/song940/node-escpos/blob/master/printer.js#L186). Maybe thermal printers differ in max characters per line. Would it be better to make line width as variable in printer options?

```js
const options = { encoding: 'GB18030', lineWidth: 32 }
const printer = new escpos.Printer(device, options)
```

song940, 2019-12-12:
Pull request are welcome.

closed 2019-12-12 by song940

KakarN, 2019-12-12:
@elixiao I am having the same issue with the printing the table alignment. I tried checking out the line of code you mentioned, but did not it in the print.js. Could you please tell me how did you fix it?