lsongdev/node-escpos #224: How to for loop and for each

artty58, 2019-09-06:
for loop and for each data to bill

Psychopoulet, 2019-09-06:
that is not very explicit...
any case to submit ?

artty58, 2019-09-06:
I want to print receipt in many items but it's not fixed then I think to use "for" or "foreach" for printing any item. Can you help me?

artty58, 2019-09-06:
`for(var i; i<dataReceipt.length;i++){
                    .text("1. " + dataReceipt[i].tracking + " " + dataReceipt[i].province_name + " " + dataReceipt[i].zipcode)
                    .text('   ' + 'ชื่อผู้รับ : ' + dataReceipt[i].receiver_name)
                    .text('   ' + 'พัสดุธรรมดา ' + dataReceipt[i].alias_size + '/' + dataReceipt[i].parcel_type + "                " + dataReceipt[i].size_price + " บาท")
                    }`

artty58, 2019-09-06:
[image]

artty58, 2019-09-06:
[image]

artty58, 2019-09-06:
@Psychopoulet @song940

Psychopoulet, 2019-09-09:
... maybe because you try to use a function of an object without writing the object (".text()") ??
there is no issue with the package, only with the code ><
just read the debug stack...

closed 2019-09-09 by Psychopoulet