lsongdev/node-escpos #229: ESCPOS not working on Electron app

Bhanuka94, 2019-09-19:
The project built with Electron and ESCPOS library.

When clicking the "Print" button on electron window, the printer should print receipt and open cashdrawer.

But this is not happening.

When running "npm start"  the electron window(index.html) opens. The index.html print button brings only the alert window. But rest of the ESCPOS functions are not working.

When seperately running the "renderer.js" file in terminal by running "node renderer.js", the printer and cashdrawer works. The alert("Cash Drawer Open");  statement should be commented.

The problem is when clicking the print button in the window, only alert works. Other ESCPOS functions are not working.

****

Please note I have installed electron(latest version) with "npm install --save-dev electron" command. 
After that I added all "ESCPOS" library modules(https://github.com/song940/node-escpos) to a new folder called "escpos" inside node modules.  

My GitHub repo is given here. (https://github.com/Bhanuka94/electronCash).

Psychopoulet, 2019-09-19:
did you try to install escpos dependencies ?
with a proper install of the module ("npm install escpos") instead of git download

artty58, 2019-09-20:
[image]

artty58, 2019-09-20:
yes , i do npm install escpos

Bhanuka94, 2019-09-20:
> did you try to install escpos dependencies ?
> with a proper install of the module ("npm install escpos") instead of git download

Yes. I did install dependencies.
[image]

artty58, 2019-09-20:
I would like to download your project.

artty58, 2019-09-20:
[image]
How to

artty58, 2019-09-20:
[image]

Bhanuka94, 2019-09-20:
> I would like to download your project.

My GitHub repo is given here. (https://github.com/Bhanuka94/electronCash).

Once you open this with VS Code, install the following dependencies via terminal.
 
**npm install --save-dev electron

npm i escpos**

Afterwards, run the project "npm start".
/////////////////////////
Please note the system is not working well yet. I am still trying to open printer with window "print" button.

Bhanuka94, 2019-09-20:
> did you try to install escpos dependencies ?
> with a proper install of the module ("npm install escpos") instead of git download

I tried some exception handling with try, catch statements to find the actual error. After clicking the print button, I got an error.

I got the error as "**Cannot access 'device' before initialization**'".

How to resolve this?
[image]

Bhanuka94, 2019-09-20:
> I tried some exception handling with try, catch statements to find the actual error. After clicking the print button, I got an error.
>
> I got the error as "**Cannot access 'device' before [...]

This is the code in renderer.js.
[image]

Bhanuka94, 2019-09-20:
> [image]
> How to

You have to install the following dependencies via terminal.

npm install --save-dev electron
npm i escpos

Psychopoulet, 2019-09-20:
why the f*ck is there two different conversations with two different people with two different projects ??

Psychopoulet, 2019-09-20:
i will delete all the artty58's posts, who is certainly speaking of an other thread

Psychopoulet, 2019-09-20:
you cannot use escpos in a client way. it's a server-side package which not works on browsers

closed 2019-09-20 by song940

artty58, 2019-09-21:
ปากดีหรอไอสัส