lsongdev/node-escpos #232: print on windows shared printer

everton-e26, 2019-10-05:
How could i print on a shared printer connected to another Windows pc?

i've tried this
`const device = new escpos.Network("\\\\192.168.0.106\\sweda");`

the escpos.Network only support printer connected directly to the net?

m4heshd, 2019-10-06:
Wondering the same thing. Ends up with
```
events.js:112
      throw er; // Unhandled 'error' event
      ^

TypeError: Cannot read property 'write' of null
    at Network.write (Node\dot-matrix-test\node_modules\escpos\adapter\network.js:45:15)
    at Printer.flush (Node\dot-matrix-test\node_modules\escpos\printer.js:717:16)
    at Printer.close (Node\dot-matrix-test\node_modules\escpos\printer.js:742:15)
```

Psychopoulet, 2019-10-07:
it's a network protocole, not a windows one.
you juste have to set the IP printer adress, it works with any OS

Psychopoulet, 2019-10-07:
but you have to put the printer on your network, with a static IP

m4heshd, 2019-10-07:
I'm switching from JAVA and i wish it was possible to print to a file port just like the [RTextPrinter](http://www.java4less.com/textprinter/index.html).

Psychopoulet, 2019-10-08:
this is a specific protocol, it's possible to use it, but not with the Network adapter, you'll have to create your own one

Psychopoulet, 2019-10-08:
BTW, the RTextPrinter seems very limited compared to "printer" and "escpos" npm modules...

closed 2019-10-10 by Psychopoulet