lsongdev/node-escpos #236: Examples/image.js not working

naphattharawat, 2019-11-12:
I clone source code from github and test print "examples/image.js" is not working

It error TypeError: printer.align(...).image(...).image is not a function
[image]

DieserMerlin, 2019-12-02:
This is because the "image" function is asynchronous and returns a promise. So instead of
```javascript
printer.text("Hey").image(img).cut().close();
```
you need to do something similar to this:
```javascript
printer.text("Hey").image(img).then(() => printer.cut().close());

// Another example using an async-function:
const foo = async () => {
    (await printer.text("Hey").image(img)).cut().close();
};
```

You can also use raster instead of image which is synchronous and doesn't return a promise, but this way I cannot align the image very well.

```javascript
printer.raster(img).cut().close();
```

592da, 2019-12-07:
there is another issue.
I am using raster to print images, but in case the image is too big (byte length) the printer prints binary code... someone knows how to fix it?

closed 2020-06-16 by song940