lsongdev/node-escpos #238: Can someone recommend me a printer?

dolanmiu, 2019-11-18:
Which printers do you guys use for Linux and Windows?

Have you guys tried USB and Network printers? Which ones work?

Psychopoulet, 2019-11-19:
I use this, it's my main (but in black)
https://www.epson.fr/viewcon/corporatesite/products/mainunits/overview/9450

dolanmiu, 2019-11-19:
do you use this with Windows or Linux?

Psychopoulet, 2019-11-20:
Both Linux and Windows.
Like said on the doc, you need a driver (Zadig) on Windows if you use it in a USB way.
You don't with web, serial or bluetooth.

dolanmiu, 2019-11-20:
I am thinking of connecting the printer to ethernet (LAN)

and let a linux machine and a windows machine print from it

I don't need any drivers? Just this library?

Psychopoulet, 2019-11-21:
yes, the network adapter manage the connection for you, you just have to set the right parameters (ip, etc...)

closed 2019-11-21 by Psychopoulet