lsongdev/node-escpos #241: How to print the text from the next line, if the word extends the paper?

KakarN, 2019-12-11:
I am working with the default example and its working fine. The only thing I have changed is that I have increased the length of english text to be printed.

```
device.open(function(){
  printer
  .font('a')
  .align('ct')
  .style('bu')
  .size(1, 1)
  .text('"The quick brown fox jumps over the lazy dog" is an English-language pangram—a sentence that contains all of the letters of the alphabet.')
  .text('敏捷的棕色狐狸跳过懒狗')
  .barcode('1234567', 'EAN8')
  .table(["One", "Two", "Three"])
  .tableCustom([ 
    { text:"Left", align:"LEFT", width:0.33 },
    { text:"Center", align:"CENTER", width:0.33},
    { text:"Right", align:"RIGHT", width:0.33 }
  ]) 
  .qrimage('https://github.com/song940/node-escpos', function(err){
    this.cut();
    this.close();
  });
});
```

Here is the photo of the print:

[image]

As you can see the words are broken on the right side if it extends the paper. My issue is with how the word breaks automatically to print the text. It seems like the css `word-break` property is at `break-all` mode. 

Is there any way to change this behaviour so that instead of breaking the word, it will start printing from the next line?

Psychopoulet, 2019-12-12:
https://github.com/song940/node-escpos/issues/223#issuecomment-564814052

SmirnovM91, 2020-02-20:
Hello, Do you have any ideas how to fix it?

eliseudr, 2021-03-17:
Try using a smaller font size .size(0.01, 001) and it seems that your paper size is smaller then the default.
You can try aswell doing a .drawLine() before de .barCode