lsongdev/node-escpos #245: Encode picture

Nehved, 2019-12-17:
Hi there. I have some troubles with printing images. Here my source code
`const device = new escpos.Network(IP);
const options = { encoding: 'CP860' };
const printer = new escpos.Printer(device, options);
const tux = path.join(__dirname, 'tux.png');

escpos.Image.load(tux, (image) => {
  device.open(() => {
    printer.text('hello here');
    printer.image(image);
    printer.cut();
    printer.close();
  });
});`

below I attached photo of result, thanks for your help)

[image]

Nehved, 2019-12-17:
Btw, I use Ubuntu 18.04 + Electron app, printer HP.

Psychopoulet, 2019-12-18:
printer.image is an async method

Nehved, 2019-12-18:
@Psychopoulet thanks for response, I refactored my code in next way: 
```
const { Printer, Network, Image } = require('escpos');
const device = new Network(ip);
const printer = new Printer(device);
const tux = path.join(__dirname, 'tux.png');

Image.load(tux, (image) => {
  device.open(async () => {
    await printer.image(image);
    printer.cut().close();
  });
});
```
result still the same, could you pls provide working example?

Psychopoulet, 2019-12-19:
... in your electron app do you use this package in a front way ??

Nehved, 2019-12-19:
@Psychopoulet no, it is located in my main file for backend part, like "electron.js"

Psychopoulet, 2019-12-20:
what is your printer's model ?
are you sure thate escpos standard is enable on it ?