lsongdev/node-escpos #249: Error: Can't close device with a pending request

nicolasbadia, 2020-01-10:
I add the exception 'Error: Can't close device with a pending request' each time I close a printer used with the USB adaptor.
Removing this line solved my issue. I believe the line above already take care of closing the connection.

Psychopoulet, 2020-01-13:
"Can't close device with a pending request" => this probably happens because you're trying to close the device during a pending request....
This code works, you should controls the requests you make before and ensure the ending of them (some of them are probably asynchronous, like printing pictures)

nicolasbadia, 2020-01-13:
The exception is caused by this simple code :

```
var adapter = new escpos.USB(),
  printer = new escpos.Printer(adapter),
  isConnected = null;

adapter.open(function(err) {
  if (err) isConnected = false;
  else {
    isConnected = true;
    printer.close();
  }
});
```

nicolasbadia, 2020-01-13:
I thought about it again and it may be due to concurrent requests. I will do some tests.

nicolasbadia, 2020-01-13:
No, it is not due to concurrent requests...
The first close on the USB adaptor works fine, but if I open and close it again (4 seconds later), the error occurs.
Can't the pending request be a close request? Though, 4 seconds should be enough to close it...

nicolasbadia, 2020-02-22:
I found the cause of the problem.

The close command may actually take some time, and if you open the connection while another connection is not yet close, it crashes (even with try/catch). I found out the first argument of the close method is a callback which is called when the connection is fully close. With it, you can easily delay new connections until the printer is ready.

It could be nice to add a queue system to the USB adaptor to handle this edge case. But if you think it is out of the scope of this library, feel free to close the issue.

closed 2022-08-04 by song940