lsongdev/node-escpos #260: install only Network variant?

Spider84, 2020-02-17:
Hello. Can I install only Network version w/o serial or usb? For preventing building backend libraries?
like npm i escpos.network --save?

Psychopoulet, 2020-02-17:
nop, but you can download the package with git, without build it ?
IDK if npm have an option to avoid build step while downloading

song940, 2020-02-24:
https://github.com/song940/node-escpos/issues/264

closed 2020-02-24 by song940

vgomezmza, 2020-11-27:
Good day someone could solve this error:

TypeError: escpos.Network is not a constructor
 at Object.<anonymous> (C:\wamp\www\controlFiscal2\app.js:6:17)
    at Module._compile (internal/modules/cjs/loader.js:1137:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1157:10)
    at Module.load (internal/modules/cjs/loader.js:985:32)
    at Function.Module._load (internal/modules/cjs/loader.js:878:14)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:71:12)
    at internal/main/run_main_module.js:17:47

const escpos = require('escpos');
// install escpos-usb adapter module manually
escpos.USB = require('escpos-usb');
// Select the adapter based on your printer type
//const device  = new escpos.USB();
const device  = new escpos.Network('localhost',9100);
// const device  = new escpos.Serial('/dev/usb/lp0');
 
const options = { encoding: "GB18030" /* default */ }
// encoding is optional
.
.
.