lsongdev/node-escpos #264: escpos@v3

song940, 2020-02-24:
Hello everyone, 

Today, we published first alpha version of v3. A BREAKING CHANGE.

I move adapter out of escpos the main package to `escpos-*` package.

Welcome fork + commit and send pull-request to this repo. I'll review and merge it ASAP.

Thanks.

g3rrydanc3, 2020-02-24:
Why the version back to 0.x.x?
This can led confusing for package manager
If it breaking changes, then do major versioning.

Please follow semantic versioning
https://devhints.io/semver

bigrocs, 2020-02-25:
Is there a tutorial for using v3

bigrocs, 2020-02-25:
v3
Cannot find module './printer'
Printer.Printer = require('./printer');

[image]

song940, 2020-02-25:
@g3rrydanc3 Oh, my mistake, now it's correct.
@bigrocs I updated [README file](https://github.com/song940/node-escpos#example). And it will show you how to use in v3 version.

chourouk-mrad, 2020-02-25:
@song940 

When i try to connect to the printer via network. 

`const connection = new escpos.Network(address);`

I get the error **Unhandled Rejection (TypeError): net.Socket is not a constructor**

Do you have an idea if it's related to the new released version or it's something else ?

_NB: my printer is EPSON TM-U220_

gpallas16, 2020-02-25:
When I run npm i escpos-bluetooth runs automatically node-gyp configure build and then I get this error:

ELIFECYCLE
npm ERR! errno 1
npm ERR! node-bluetooth@1.2.6 install: `node-gyp configure build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the node-bluetooth@1.2.6 install script.

Should I install or configure anything else before installing the bluetooth adapter?

My current setup:
   Windows 10
   Electron 7.1.5
   Node 12.16.1
   Npm 6.13.4

bigrocs, 2020-02-26:
@gpallas16  I encountered a problem while building node-bluetooth

song940, 2020-02-26:
@gpallas16 Please referred to [node-bluetooth](https://github.com/song940/node-bluetooth) .

song940, 2020-02-26:
@chourouk-mrad what version of Node.js ?

sudippaudel793, 2020-04-07:
> @gpallas16 Please referred to [node-bluetooth](https://github.com/song940/node-bluetooth) .

same problem occured with mac os. i have node-gyp of version v6.1.0 but also escpos-bluetooth is not installed and shows the error below:-
[image]

yelhouti, 2021-01-28:
Since we are moving to v3 and already have breaking changes, can we just use async await instead of all the callbacks.
For example open could just return a Promise that fails when open fails...

ehmad, 2021-08-03:
> @song940
> 
> 当我尝试通过网络连接到打印机时。
> 
> `const connection = new escpos.Network(address);`
> 
> 我收到错误**Unhandled Rejection (TypeError): net.Socket is not a constructor**
> 
> 你知道它是否与新发布的版本有关还是其他什么？
> 
> _注意：我的打印机是 EPSON TM-U220_

I also encountered the same problem. Did you solve it?

ehmad, 2021-08-03:
> @chourouk-mrad什么版本的 Node.js？

I also encountered the same problem.
node:14.17.0

ehmad, 2021-08-03:
> @ chourouk-mrad
> 
> When i try to connect to the printer via network.
> 
> `const connection = new escpos.Network(address);`
> 
> I get the error **Unhandled Rejection (TypeError): net.Socket is not a constructor**
> 
> Do you have an idea if it's related to the new released version or it's something else ?
> 
> _NB: my printer is EPSON TM-U220_

I also encountered the same problem.Did you solve it?

FawzyMokhtar, 2022-01-14:
Can we use this package in Ionic 6 to support printing on Android & IOS ? @song940

prashanta-sam-velectico, 2024-03-01:
> I also encountered the same problem.Did you solve it?

>> I am getting  this error  "escpos.Network is not a constructor"

prashanta-sam-velectico, 2024-03-01:
I am getting this error "escpos.Network is not a constructor"

closed 2024-12-17 by lsongdev