lsongdev/node-escpos #272: Can't resolve 'escpos-bluetooth'

sudippaudel793, 2020-03-14:
what to do with this???

song940, 2020-03-15:
https://www.npmjs.com/package/escpos-bluetooth

closed 2020-03-15 by song940

sudippaudel793, 2020-03-15:
[image]
This issues dont let me move forward

paul122297, 2020-04-21:
Same issue here using Ubuntu and Electron Vue

paul122297, 2020-04-21:
Same issue here using Electron and Vue

`
[image]
[image]

`