lsongdev/node-escpos #277: can mac use gprinter by usb

zhyc9de, 2020-04-02:
i can get device used `device  = new escpos.USB()`
but when i run example, nothing happend

Psychopoulet, 2020-04-02:
did you install the Zadig driver ?

zhyc9de, 2020-04-02:
> did you install the Zadig driver ?

i thought Zadig just run in windows platform?
but i really install something like libusb

song940, 2020-04-03:
@zhyc9de use [lsusb](https://gist.github.com/song940/7149576) command to see what pid and vid of your usb printer.

zhyc9de, 2020-04-03:
> @zhyc9de use [lsusb](https://gist.github.com/song940/7149576) command to see what pid and vid of your usb printer.

```
▶ bash lsusb.sh
2020-04-03 19:08:28.024 system_profiler[39912:4103234] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2020-04-03 19:08:28.025 system_profiler[39912:4103234] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2020-04-03 19:08:28.025 system_profiler[39912:4103234] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2020-04-03 19:08:28.025 system_profiler[39912:4103234] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2020-04-03 19:08:28.030 system_profiler[39912:4103234] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2020-04-03 19:08:28.030 system_profiler[39912:4103234] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
Bus  Device 1: ID 05e3:0610 (Genesys Logic, Inc.) USB2.1 Hub
Bus  Device 3: ID 0471:0055 (Philips Consumer Lifestyle BV) Gprinter USB Printer  Serial: 0002A0000000
Bus  Device 2: ID 1d5c:7102 (Fresco Logic Inc.) Generic Billboard Device
Bus 00 Device 1: ID 05e3:0626 (Genesys Logic, Inc.) USB3.1 Hub
Bus 00 Device 2: ID 0bda:8153 (Realtek Semiconductor Corp.) USB 10/100/1000 LAN  Serial: 001000001
Bus 00 Device: ID 05ac:8102 (Apple Inc.) Touch Bar Backlight  Serial: 0000000000000000
Bus 00 Device: ID 05ac:8302 (Apple Inc.) Touch Bar Display  Serial: 0000000000000000
Bus 00 Device 2: ID 05ac:027c (Apple Inc.) Apple Internal Keyboard / Trackpad  Serial: FM79217010LJ2KVAP+TDZ
Bus 00 Device: ID 05ac:8103 (Apple Inc.) Headset  Serial: 000000000000
Bus 00 Device: ID 05ac:8262 (Apple Inc.) Ambient Light Sensor  Serial: 000000000000
Bus 00 Device: ID 05ac:8514 (Apple Inc.) FaceTime HD Camera (Built-in)  Serial: CC29173PZ5GJ3Y338
Bus 00 Device: ID 05ac:8233 (Apple Inc.) Apple T2 Controller  Serial: 0000000000000000
```

zhyc9de, 2020-04-03:
```
const device = new escpos.USB();
console.log(device)
```
result: 
```
▶ node examples/index.js
USB {
  _events: [Object: null prototype] {},
  _eventsCount: 0,
  _maxListeners: undefined,
  device: Device {
    busNumber: 20,
    deviceAddress: 3,
    deviceDescriptor: {
      bLength: 18,
      bDescriptorType: 1,
      bcdUSB: 512,
      bDeviceClass: 0,
      bDeviceSubClass: 0,
      bDeviceProtocol: 0,
      bMaxPacketSize0: 64,
      idVendor: 1137,
      idProduct: 85,
      bcdDevice: 256,
      iManufacturer: 1,
      iProduct: 2,
      iSerialNumber: 3,
      bNumConfigurations: 1
    },
    portNumbers: [ 3, 2 ],
    _configDescriptor: {
      bLength: 9,
      bDescriptorType: 2,
      wTotalLength: 32,
      bNumInterfaces: 1,
      bConfigurationValue: 1,
      iConfiguration: 0,
      bmAttributes: 160,
      bMaxPower: 50,
      extra: <Buffer >,
      interfaces: [Array]
    }
  },
  [Symbol(kCapture)]: false
}
```

zhyc9de, 2020-04-03:
maybe my printer not support esc mode...

closed 2020-04-03 by zhyc9de