lsongdev/node-escpos #280: Exception: __webpack_require__(...) is not a function

infinitetrinity, 2020-05-03:
Using Electron and VueJS will result to this error when importing for use.
Look like it's an issue with iconv-lite when use for electron app.
Please check these links for more details as well.

https://github.com/ashtuchkin/iconv-lite/issues/205#issue-370448101
https://github.com/Klemen1337/node-thermal-printer/issues/109#issuecomment-558490109

```
`Uncaught TypeError: __webpack_require__(...) is not a function
    at eval (index.js?acf9:144)
    at Object../node_modules/iconv-lite/lib/index.js (chunk-vendors.js:2796)
    at __webpack_require__ (app.js:785)
    at fn (app.js:151)
    at eval (index.js?eb39:4)
    at Object../node_modules/escpos/index.js (chunk-vendors.js:2494)
    at __webpack_require__ (app.js:785)
    at fn (app.js:151)
    at eval (receipt.js:5)
    at Module../src/utils/receipt.js (app.js:4332)`
```

Psychopoulet, 2020-05-03:
I guess you use it in the front part of your electron app (with view ?).
it's designed to work with node, in a back-end integration. You can do it in the back part of your app, by integrate some micro-API for example

closed 2020-06-16 by song940

stan-sack, 2020-12-12:
it works in electron. you just need to stop it being transpiled for a browser. fix is [here](https://github.com/ashtuchkin/iconv-lite/issues/204#issuecomment-432048618)