lsongdev/node-escpos #294: Remove the default font size for printer.size()

encodecafe, 2020-06-15:
Font size is not working when we use values ​​like 0.0 | 0.1 | 1.0 | 1.1.
So, why not use values ​​between 0 and 7 to set the width and height values ​​for the font size?
I propose to remove the condition that defines this behavior and to discern a limit between 0 and 7 for width and height.
What do you think guys?

ngmiduc, 2020-12-26:
I also having troubles with the font size. It is not really easy to adjust the size.

encodecafe, 2020-12-26:
Hi ngmiduc!  You just set the values between 0 and 7 to printer.size(0,0) ~ printer.size(7,7). The variations in values change the font size. You can set (0,1), (1,0), (1,1), (3,1) ..., any combination.

ngmiduc, 2020-12-26:
Ah okay. In the Docs it is written for the values 1 and 0 only.

Okay good to know.

closed 2022-08-04 by song940