lsongdev/node-escpos #297: Statuses.js error "./node_modules/escpos/statuses.js"

kyawswar, 2020-06-26:
I used with vuejs. I face the error below.
Module parse failed: Unexpected token (7:7)
You may need an appropriate loader to handle this file type, currently no loaders are configured to process this file. See https://webpack.js.org/concepts#loaders
|
| class DeviceStatus {
>   byte = '';
|   bits = [];
|   bitsAsc = [];

 @ ./node_modules/escpos/index.js 12:17-38
 @ ./node_modules/cache-loader/dist/cjs.js??ref--12-0!./node_modules/babel-loader/lib!./node_modules/cache-loader/dist/cjs.js??ref--0-0!./node_modules/vue-loader/lib??vue-loader-options!./src/App.vue?vue&type=script&lang=js&
 @ ./src/App.vue?vue&type=script&lang=js&
 @ ./src/App.vue
 @ ./src/main.js
 @ multi (webpack)-dev-server/client?http://192.168.100.9:8080&sockPath=/sockjs-node (webpack)/hot/dev-server.js ./src/main.js

How can i fix that? please instruct me. thanks

[image]

kyawswarthank, 2020-06-29:
### Operation is OK !!
I setup with below link.
https://medium.com/@nyorikakar/printing-with-vue-electron-and-node-f52a730e63e7

I face the above issue. 
I fixed that below.
Step 1:  All code need to place at 'background.js'.
````js
....
const escpos = require('escpos');
escpos.USB = require('escpos-usb');
const { ipcMain } = require('electron')
......
......
ipcMain.on('print', (event, arg) => {
// console.log(arg) // receive data from arg
  print(arg);
})
const print = (arg) => {

  setTimeout(() => {
    console.log('test print start')
  
    // Select the adapter based on your printer type
    const device  = new escpos.USB('0x0fe6','0x811e');
  // const device  = new escpos.Network('localhost');
  // const device  = new escpos.Serial('/dev/usb/lp0');
  
    const options = { encoding: "GB18030" /* default */ }
  // encoding is optional
  
    const printer = new escpos.Printer(device, options);
  
  
  
    device.open(function(){
      printer
          .font('a')
          .align('ct')
          .style('bu')
          .size(1, 1)
          .text(arg)
          .size(.001,.001)
          .tableCustom(
            [
              { text:"Apple                         200", align:"LEFT", width:0.5},
              { text:"Orange                        200", align:"LEFT", width:0.5},
            ],
            { encoding: 'cp857', size: [2, 1] } // Optional
          )
          .qrimage('https://github.com/song940/node-escpos', function(err){
            console.log(err);
            this.cut();
            this.close();
          })
          .text('Thanks YOU!')
          ;
    });
    
    console.log('test print end')
  }, 1000)

};
````
Step 2: Update your App.Vue file like that.
````js
<template>
    <div id="app">
        <h3>POS Title</h3>
        <table style="width: 100%">
          <tr>
            <th>Item</th>
            <th>Price</th>
          </tr>
          <tr>
            <td>Apple </td>
            <td>200</td>
          </tr>
          <tr>
            <td>Orange </td>
            <td>200</td>
          </tr>
          <tr>
            <td>Total </td>
            <td>400</td>
          </tr>
        </table>
        <br>
        <button @click="printPaper" id="print">Print</button>
    </div>
</template>

<script>
    // import escpos from 'escpos'
    // escpos.USB = require('escpos-usb');
    const { ipcRenderer } = require('electron');
    export default {
        name: 'app',
        methods: {
            printPaper() {
                console.log('printPaper! start')
                ipcRenderer.send('print','your data go here')
                console.log('printPaper! end')
            }
        }
    }
</script>

<style>
#app {
  font-family: Avenir, Helvetica, Arial, sans-serif;
  -webkit-font-smoothing: antialiased;
  -moz-osx-font-smoothing: grayscale;
  text-align: center;
  color: #2c3e50;
  margin-top: 60px;
}
</style>

````

Psychopoulet, 2020-06-30:
the hook you've made is exactly what you have to do : this module is not build to work in a front way, so you have to make it work with a communication client-server and run it server-side with node

closed 2020-06-30 by Psychopoulet

erickcg, 2020-07-16:
Just had this problem. I have always used this package with vue-electron and this last version has this issue (I think it has to do with the target node version but i am not sure).

Anyway. I installed "escpos": "^2.5.2", and everything is fine. just ran electron-rebuild and its ready to print.

by-iankeezy, 2020-08-21:
Hey @kyawswar i have tried your method and it failed, please check out my issue #306 (I have attached images)

juanzeta33, 2020-12-02:
@kyawswar Hi, I still having this Issue, I try your solution but dont work for me :(

Any idea or any code to help me?

isakulaksiz, 2022-03-20:
`sed '7,10d' node_modules/escpos/statuses.js`
You can use bash script