lsongdev/node-escpos #301: Cannot get Printer Status

whizyrel, 2020-06-30:
Good day,
So, I use this version || the latest version => 3.0.0-alpha.4 and i tried get Printer statuses using the snippet below and this is the error i got.
`
// snippet
printer
        .getStatuses(statuses => {
            statuses.forEach(status => {
                console.log(status.toJSON());
            })
        })
        .close();
`

`
// error
TypeError: this.adapter.read is not a function
      at Printer.getStatuses (/home/whizyrel/Documents/Code/JS/lp-sales/node_modules/escpos/index.js:894:16)
      at /home/whizyrel/Documents/Code/JS/lp-sales/handlers/printer.ipc.js:44:16
      at Interface.<anonymous> (/home/whizyrel/Documents/Code/JS/lp-sales/node_modules/escpos-usb/index.js:134:25)
      at Device.<anonymous> (/home/whizyrel/Documents/Code/JS/lp-sales/node_modules/usb/usb.js:341:6)
`

mln4you, 2020-08-17:
Same problem here

whizyrel, 2020-10-02:
This module requires attention

closed 2020-10-02 by whizyrel