lsongdev/node-escpos #302: German Encoding

mats-codes, 2020-07-06:
Hey,

first of all thanks for the awesome project! Since I'm from germany I'd need letter like ä, ö, ü, ß, &. Is there any documented overview of possible encodingd?

Does anybody know, which I can pick?

Thanks and Regards

mats-codes, 2020-07-06:
Found the solution:
{
  encoding: 'Cp850',
}

closed 2020-07-06 by mats-codes

oliverdotbecker, 2020-11-07:
Correct. Change this in printer.js in version 2.4:

`function Printer(adapter) {
  if (!(this instanceof Printer)) {
    return new Printer(adapter);
  }
  var self = this;
  EventEmitter.call(this);
  this.adapter = adapter;
  this.buffer = new Buffer();
  this.encoding = 'Cp850';
};`

ngmiduc, 2021-01-10:
Hi!
Somehow I have added "cp850" or "Cp850" and the ä, ö, ü ... are still malformatted.

I init encoding with 

`const printer = new escpos.Printer(device, { encoding: 'Cp850' })`