lsongdev/node-escpos #303: cashdraw() issue

ChinLii, 2020-07-18:
When I call cashdraw() method the first print is fine but the next print gonna have extra character at the top of paper.

How can I fix this case?  
Are they any way to open the cashdrawer without having the extra character ? 

The code that I test it on is 
`networkDevice.open(function(error){
       printer.align('ct').text('test').cashdraw().cut('partial').close()
  });`

Here are the picture of receipt that I mentioned about 
[https://drive.google.com/file/d/1kDmbVDdTDzgqr8RzqGSwJV02rWUZdb-I/view?usp=sharing](url)

zhaofenghao, 2020-09-09:
Same problem

And I call cashdraw() two times, the cashdraw can only open one time. why?

zhaofenghao, 2020-09-11:
@ChinLii 
I found a way! 
You need modify cashdraw() like this.  And there will be no extra characters on the next print!

``
/**
 * [function Send pulse to kick the cash drawer]
 * @param  {[type]} pin [description]
 * @return {[Printer]} printer  [the escpos printer instance]
 */
Printer.prototype.cashdraw = function (pin) {
  this.buffer.write(_.CASH_DRAWER[
    'CD_KICK_' + (pin || 2)
  ]);
  var buf = this.buffer.flush();
  this.adapter.write(buf, callback);
  return this;
};
``

zhaofenghao, 2020-09-11:
done!
i modified escpos/index.js

Printer.prototype.init = function (pin) {
  this.buffer.write(_.HARDWARE[
    'HW_INIT'
  ]);
  return this;
};

Before print, you need call init() to clear the data in buffer.

device.open(function(err) {
  printer.init().model('qsprinter'); //
})

frankfabregat, 2021-07-26:
This may be a little late, but if you are still looking for an answer or for other having this problem, I managed to solve it by using
`printer.flush().raw('1b 70 01 10 fa').flush()` instead of `printer.cashdraw()`. The raw command sends the command to open the cash drawer directly to the printer. I do not know why the cashdraw() function prints an extra character, but this solved my issue.

kureci, 2021-11-30:
Just created a pull request that should fix this, I hope maintainer gets to check it and merge it in.

https://github.com/song940/node-escpos/pull/373