lsongdev/node-escpos #306: Statuses.js error

by-iankeezy, 2020-08-21:
Hello,
I'm encountering the issue below 

[image]

**My background.js is as follows**

[image]

I have checked #297 and tried the fix but it hasn't worked for me 

I am using this tutorial -> https://medium.com/@nyorikakar/printing-with-vue-electron-and-node-f52a730e63e7

### Environment

- yarn 1.22.4
- escpos 3.0.0-alpha.6
- escpos-usb 3.0.0-alpha.4

### Steps to reproduce:

1. Install electron builder `vue add electron-builder`
2. Install escpos `yarn add escpos`
3. Add escpos to background.js
4. Run `yarn electron:serve`

### **Full Error Stack**

`⠹  Bundling main process...`
` ERROR  Failed to compile with 1 errors                               9:02:13 AM`
 `error  in ./node_modules/escpos/statuses.js`
`Module parse failed: Unexpected token (7:7)`
`You may need an appropriate loader to handle this file type, currently no loaders are configured to process this file. See https://webpack.js.org/concepts#loaders`
`|` 
`| class DeviceStatus {`
`>   byte = '';`
`|   bits = [];`
`|   bitsAsc = [];`

 `@ ./node_modules/escpos/index.js 12:17-38`
 `@ ./src/background.js`
 `@ multi ./src/background.js`

`ERROR  Build failed with errors.`
`error Command failed with exit code 1.`
`info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.`
`

zhaofenghao, 2020-09-11:
do not support es6?  try to delete "async"

juanzeta33, 2020-12-02:
same follow steps, same issue ;?

Ian-Leo, 2020-12-07:
I had the issue despite removing async

JerryhyChen, 2020-12-13:
Did you write this code in your `vue.config.js` ?
You can try it.
```
module.exports = {
    pluginOptions: {
        electronBuilder: {
            externals: ['escpos', 'escpos-usb'],
        }
    }
}
```

Ian-Leo, 2020-12-13:
> Did you write this code in your `vue.config.js` ?
> You can try it.
> ```
> module.exports = {
>     pluginOptions: {
>         electronBuilder: {
>             externals: ['escpos', 'escpos-usb'],
>         }
>     }
> }
> ```

Will try it

juanzeta33, 2020-12-14:
Hi guys @Ian-Leo,

I fix it by doing the following:

1. delete node_modules and run npm install
2. adding tthe lines that's mentions @JerryhyChen 
```
module.exports = {
    pluginOptions: {
        electronBuilder: {
            externals: ['escpos', 'escpos-usb'],
        }
    }
}
```
3. Adding print code in background.js and calling the event for print from my .vue

Y2K222, 2021-01-11:
Removing these lines

 ```
byte = '';
 bits = [];
 bitsAsc = [];
```
in node_modules/escpos/statuses.js 
Works for me

Ian-Leo, 2021-01-31:
> Hi guys @Ian-Leo,
> 
> I fix it by doing the following:
> 
> 1. delete node_modules and run npm install
> 2. adding tthe lines that's mentions @JerryhyChen
> 
> ```
> module.exports = {
>     pluginOptions: {
>         electronBuilder: {
>             externals: ['escpos', 'escpos-usb'],
>         }
>     }
> }
> ```
> 
> 1. Adding print code in background.js and calling the event for print from my .vue

This worked and now i can print from background.js but when inside a vue page i cannot. I still get the same error.

yelhouti, 2023-03-26:
This issue seems to happen only windows, deleting the line as mentioned seems to solve the issue for the devs that have it. I couldn't figure out why its different between linux and windows.