lsongdev/node-escpos #308: Network Printer Syntax error

PemmasaniRajesh, 2020-09-05:
I've installed the escpos library in ElectronJS. But when I'm trying to runt the project I'm getting the following error.

Node Version: 12.13.1

App threw an error during load
TypeError: escpos.Network is not a constructor
    at Object.<anonymous> (C:\Work\electron\electron-app\main.js:9:23)
    at Module._compile (internal/modules/cjs/loader.js:1145:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1166:10)
    at Module.load (internal/modules/cjs/loader.js:981:32)
    at Module._load (internal/modules/cjs/loader.js:881:14)
    at Function.Module._load (electron/js2c/asar.js:769:28)
    at loadApplicationPackage (C:\Work\electron\electron-app\node_modules\electron\dist\resources\default_app.asar\main.js:109:16)
    at Object.<anonymous> (C:\Work\electron\electron-app\node_modules\electron\dist\resources\default_app.asar\main.js:155:9)
    at Module._compile (internal/modules/cjs/loader.js:1145:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1166:10)
(node:5964) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.

FanghanHu, 2020-10-16:
escpos.Network is not a constructor, it is a factory function,
change `const networkDevice = new escpos.Network('192.168.1.40', 9100);` to 'const networkDevice = escpos.Network('192.168.1.40', 9100);' should fix it.

T-x-s, 2021-02-23:
Hello, I also encountered this problem, but I'm a novice and can't solve this problem. Do you have a solution?

anwarpro, 2021-03-04:
after installed escpos-network, my problem goes.
`yarn add escpos-network`

closed 2021-03-30 by song940