lsongdev/node-escpos #309: Print in turkish

MarvinShelbyGT, 2020-09-10:
Hi,

I can't print char from turkish langage :

`Ç Ğ I İ Ö Ş Ü ü ş ö i ı ğ ç Â â î Î Û û`

But i have this on my printer : 

[image]

Did i miss something for print this special char ?

TY..

MarvinShelbyGT, 2020-09-16:
i try to do this : 

`const options = { encoding: "iso-8859-9"} 
const printer = new escpos.Printer(device, options)`

But i always have the wrong char, i want **ı**  and i get **²**

marcoss2009, 2020-09-21:
Try with .encode('ibm857') at first line of printing script.

`device.open(function(error){
  printer
  .align('ct')
  .encode('ibm857')
  .text('The quick brown fox jumps over the lazy dog')
  .text('Ç Ğ I İ Ö Ş Ü ü ş ö i ı ğ ç Â â î Î Û û')
  .qrimage('https://github.com/song940/node-escpos', function(err){
    this.cut();
    this.close();
  });
});`