lsongdev/node-escpos #310: printer.size(2,2) error

wahyufadz, 2020-09-13:
hello thanks for the fantastic library,..
unfortunately i cannot set size from to double the size

i use epson TM-U220
this is some case that i tried
every use case was giving different result but all stil have the normal font size

printer
      .font('A')
      .align('LT')
      .style('NORMAL')
      .size(1, 2)
      .text('Text A')

result = Text A

printer
      .font('A')
      .align('LT')
      .style('NORMAL')
      .size(2, 1)
      .text('Text A')

result = !Text A

printer
      .font('A')
      .align('LT')
      .style('NORMAL')
      .size(2, 2)
      .text('Text A')

result = "Text A

stephyswe, 2021-02-16:
> hello thanks for the fantastic library,..
> unfortunately i cannot set size from to double the size
> 
> i use epson TM-U220 on linux ubuntu 20.04
> this is some case that i tried
> every use case was giving different result but all stil have the normal font size
> 
> printer
> .font('A')
> .align('LT')
> .style('NORMAL')
> .size(1, 2)
> .text('Text A')
> 
> result = Text A
> 
> printer
> .font('A')
> .align('LT')
> .style('NORMAL')
> .size(2, 1)
> .text('Text A')
> 
> result = !Text A
> 
> printer
> .font('A')
> .align('LT')
> .style('NORMAL')
> .size(2, 2)
> .text('Text A')
> 
> result = "Text A

Hi can i ask how you get the lib to work? Using windows 10

wahyufadz, 2021-03-30:
i downgrade it with previous version